bluerobotics/companion #8: Update setup script

jaxxzer, 2017-06-20:
update setup.sh to be in line with recent webui restructure.

jaxxzer, 2018-05-21:
This issue is blocking for now using stable node (10.0.0): https://github.com/nodegit/nodegit/issues/1490

mulles, 2018-10-15:
I think the  issue  nodegit/nodegit#1490  is resolved by now.
I am working on getting the setup.sh script running on  "2018-06-27-raspbian-stretch-lite" on hardware raspi 3+. 
I hope that the image which comes out will run on all of the different raspi's ?

mulles, 2018-10-15:
Do we need nodejs-legacy  at all ? 
 It might be that sudo apt-get install-y npm would be sufficient as it will contain nodejs as well. I just started to familiarize myself with Nodejs, so please excuse if this is an obvious question.

jaxxzer, 2018-11-12:
> This issue is blocking for now using stable node (10.0.0): [nodegit/nodegit#1490](https://github.com/nodegit/nodegit/issues/1490)

This has been merged! I'll take another stab at the updates when I can.

jaxxzer, 2018-11-12:
> Do we need nodejs-legacy at all ?

@mulles I believe we do need it, but I do not remember why. I remember trying to get around it.

mulles, 2018-11-12:
I will try to run 0.0.16 on raspi 3B+. If it does not run out of the box, will see where the install script fails. I remember it was related to nodejs not running on the hardware Raspi 3B+ CPU.

jaxxzer, 2018-11-12:
The install script is broken, and you should avoid it. Thus the issue here.

mulles, 2018-11-12:
Thx for clarifying. So all images like 0.0.15 or 0.0.16 are manually produced by installing everything manually on a raspberry pi 3B ? 
Would be nice, I you could clarify who current raspi images are produced ;)

jaxxzer, 2018-11-12:
Yes, they are manually produced, this is the problem to solve.

mulles, 2018-11-12:
still for the 0.0.15 image, the setup script seems to have been used, judged by the .bash_history of the image: 
`
wget https://raw.githubusercontent.com/bluerobotics/companion/master/scripts/setup.sh
chmod +x setup.sh
./setup.sh gst
companion/tools/gps-replay.py 
ls companion/tools 
companion/tools/u
companion/tools/underwater-gps.py 
`
I am happy to help resolve any problems with the setup script.

jaxxzer, 2018-12-28:
This one is holding https://github.com/bellbind/node-v4l2camera/pull/45#issuecomment-399389412
I believe this is the last thing to resolve.

Dr-Ezzeddine, 2019-02-16:
> I am working on getting the setup.sh script running on "2018-06-27-raspbian-stretch-lite" on hardware raspi 3+.

Hi, any news about image that works on raspi 3b+? or other issues for that hardware