bluerobotics/companion #106: Add support for RPi 3 Model B+

kklemens, 2018-03-23:
Update image to support RPi 3 Model B+ hardware.

Software update was attempted through terminal on 3/22/18 using "update" and "upgrade" without success.

espresso0000, 2019-08-10:
Any update on RPi 3 B+?  Getting ready to order my parts to build, but wanted to know if I still had to use a B or if it will now work on 3B+ or even 4B?

kklemens, 2019-08-12:
@espresso0000 No further updates. Please use a 3B until additional support is announced. There is no ETA for 3B+ or 4B support at this time.

espresso0000, 2019-08-12:
Thank you for the response.  Appreciate it.  I do have several 3B+ and 4B sitting around if anything I could do to help build one just let me know.  thanks again.

jaxxzer, 2019-12-17:
See this pull request for work on an installer script: https://github.com/bluerobotics/companion/pull/284

Renegade85, 2020-02-05:
Hi guys,

this is a bit off topic, but I haven't found a better place to ask. Is there a list of supported raspberry pi's?

Thank you very much,
Renegade

Tekunalogy, 2021-11-06:
any update on this issue?

ES-Alexander, 2022-04-12:
Companion development has stopped, in favour of [BlueOS](https://discuss.bluerobotics.com/t/new-name-companion-1-0-blueos/11488), which is a ground-up rewrite, and is compatible with both Raspberry Pi 3B and 4B by default :-)