bluerobotics/companion #120: Command shortcut/script to exit all screens at once

mariyachanniwala, 2018-05-14:


mariyachanniwala, 2018-05-15:
The following two commands kill the screens but if we try to restart them by running /etc/rc.local it doesnt turn on the screen for mavproxy. I don't know why. Then if we reboot and run rc.local, all 9 screens show up.

1. killall screen 
2. screen -ls | grep pts | cut -d. -f1 | awk '{print $1}' | xargs kill

The following command seems to be doing the job right. It kills all the detached screens and then when we run rc.local, all 9 screens show up again including mavproxy. No need to reboot and run.

3. screen -ls | grep Detached | cut -d. -f1 | awk '{print $1}' | xargs kill

Any of these commands could just be entered in a shell script and run anytime we want. 
Please give your suggestions on this one.

jaxxzer, 2018-05-25:
We should add an alias to do this.

In the post update script add a line that says:

`echo "alias stopscreens="screen -ls | grep Detached | cut -d. -f1 | awk '{print $1}' | xargs kill"" >> ~/.bash_aliases`

Study bash aliases and output redirection in linux.

PR #131 (2018-06-19): Added command to kill all screens at once.
fix #120
closed 2018-06-19 by jaxxzer