bluerobotics/companion #225: Camera settings not saved

jaxxzer, 2019-02-08:
Report of regression in camera settings behavior on 0.0.16.

kklemens, 2019-03-12:
Confirmed from email from Trygve Kvåle Løken: 

"We tried creating a new profile today on the camera setup page, but had difficulties storing this profile with an absolute exposure. Once we pressed the "update profile", the value went back to 156 (default)."

jerabaul29, 2019-03-18:
Do you have an approximate timeline for when this will be fixed? ;)

Williangalvani, 2019-03-18:
I'm doing some testing here. It seems that the exposure control with the usb low light camera is not working properly. 
The V4l2 control shows the exposure field as "Inactive", which possibly means that it is disabled due to some other setting.

I'm able to reproduce the issue in 0.0.15 too.

jerabaul29, 2019-03-19:
@trygvekl you did some more observations if I remember well? Seems some update ways are working if using live control rather than saving profile?

trygvekl, 2019-03-19:
That is correct. When no profile is selected, one can press «opdate profile» and parameters seems to change as per set, also the exposure time.

Trygve

jaxxzer, 2019-03-26:
We may have more than one similar/related issue here. The original issue was about the resolution not saving. The pipeline, video format, and UVC settings are saved in separate files. The webui log gives some indication of the problem:
```
2017-12-12T15:46:18+0000 <log> index.js:75 (Object.<anonymous>) error loading video profiles from file { [Error: ENOENT: no such file or directory, open '/home/pi/camera-profiles']
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: '/home/pi/camera-profiles' }
2017-12-12T15:46:18+0000 <log> index.js:86 (Object.<anonymous>) error loading file { [Error: ENOENT: no such file or directory, open '/home/pi/camera-settings']
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: '/home/pi/camera-settings' }
```

jaxxzer, 2019-03-26:
Pretty sure this one is to blame for the resolution + ~/vidformat.param not updating:
https://github.com/bluerobotics/companion/commit/1f20cffc215516a1b505bde0ca5c563b6c65d952

related to this one:
https://github.com/bluerobotics/companion/commit/469d29d288022d758b43b54f5c20601f12603ad5

Stopping here for tonight.

jaxxzer, 2019-03-26:
FYI, the reason the new settings were comitted to ~/vidformat.param by the `start_video.sh` script is to verify that gstreamer can start the camera up with the new settings without issue before committing the settings to file.

PR #231 (2019-03-28): Fix camera settings not working properly.
It turned out that the issue was that the frontend was not matching the backend for v4l2 'menu' options.

This caused the frontend to display "Manual Mode" while "Aperture Priority Mode" was being used in the backend. This caused the exposure settings to not work unless you switched to "Aperture Priority Mode" and back to "Manual Mode" again, which updated the backend properly.

fix #225
PR #242 (2019-03-28): Fix handling of video format parameters
This way, if something goes wrong with the pipeline, the old file is restored.

Fix #225
closed 2019-03-28 by jaxxzer