bluerobotics/companion #233: Camera framerate is not fixed.

Williangalvani, 2019-03-20:
The issue was previously open in https://github.com/bluerobotics/qgroundcontrol/issues/217.

Discussion was started [here](https://discuss.bluerobotics.com/t/framerate-settings/3493).

The framerate can be checked by playing the video with this gstreamer pipeline:
```
gst-launch-1.0 -v udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' \
! rtph264depay \
! avdec_h264 \
! videoconvert \
! fpsdisplaysink
```

I was unable to make it fixed by playing with the camera settings.
Also while increasing exposure time gives us a low framerate, I was unable to get 30fps even with manual minimum exposure.

jaxxzer, 2019-03-20:
I think we should get some sort of live statistics in QGC and/or companion in order to explore this more under various conditions and with different cameras.

Williangalvani, 2019-05-17:
Another report:
https://discuss.bluerobotics.com/t/low-light-hd-usb-camera-lower-frame-rate-17fps-vs-30fps/5081

Williangalvani, 2019-05-17:
I think a good test will be using a simple pipeline. I suspect some uv4l settings are probably to blame

quattrinili, 2019-05-17:
Although in the video that I attached I did not a good job on keeping the camera with the same field of view, I thought that I tried under same conditions, but apparently not. I did another test with the spare camera I have, and indeed, it is the auto exposure that does set the maximum time according to the framerate set.

I did some tests, camera without any obstruction, and camera with my hands covering the field of view to make it darker, and indeed the framerate drops, leaving the auto exposure to the default setting which according to `v4l2-ctl` is 3. Setting the auto exposure to false through this command
`v4l2-ctl --set-ctrl=exposure_auto=1` works.

clydemcqueen, 2020-05-07:
I suggest adding `rtpjitterbuffer` to the test pipeline:

`gst-launch-1.0 -v udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtpjitterbuffer ! rtph264depay ! avdec_h264 ! videoconvert ! fpsdisplaysink`

The framerate still drops in low light, but in bright light `fpsdisplaysink` now displays 30fps.

Williangalvani, 2020-05-07:
That is right, but note that the rtpjitterbuffer will likely add some latency to the stream, too.

ES-Alexander, 2021-08-30:
I just checked if [`videorate`](https://gstreamer.freedesktop.org/documentation/videorate/index.html?gi-language=c) could help with this by adding it to the stream pipeline and it didn't seem to make a difference. Annoying because that seems to be the kind of functionality we want, although in this case perhaps the source and sink agree on the framerate, but the source is failing to actually meet that agreement (so maybe it doesn't apply?). 

Perhaps also worth noting that the sink/src options don't have `video/x-h264`, just `video/x-raw(ANY)` and `video/x-bayer(ANY)`, so it's possible it doesn't work for that reason.

I'm at least somewhat heartened by
>A conversion to a specific framerate can be forced by using filtered caps on the source pad.

Not sure how to add a filtered cap, or whether it's feasible to do so, but might be worth looking into as a next step.

ES-Alexander, 2021-09-14:
On an unrelated note, I had a bit more of a play and it seems plausible that this might be a gstreamer-specific issue. I tried the V4L2 stream rate testing [described here](https://wiki.kurokesu.com/books/recipes/page/v4l2#bkmrk-measure-actual-frame) and it's very consistent, either at or just above 30fps, or at the highest it can be by the set exposure level.

In comparison, when I'm using gstreamer to my surface computer the framerate in `fpsdisplaysink` is consistently lower, and seems to fluctuate a decent amount regardless of lighting (although low light still generally reduces the received framerate). Doesn't seem to be dropped packets as there are very few dropped frames (but that may be an invalid conclusion depending on how the dropped frame count is implemented).