bluerobotics/companion #243: Mavproxy doesn't come back after firmware update

jaxxzer, 2019-04-02:
On master, did a firmware update via webui, and lost telemetry after completion, though autopilot was running.

The board re-enumerated as /dev/ttyACM1 (I didn't check /dev/serial)

Resuming mavproxy said 'no link'

rebooting companion via webui, had telemetry back upon completion.

Williangalvani, 2019-04-02:
is your mavproxy.params still the old one?

jaxxzer, 2019-04-02:
> is your mavproxy.params still the old one?

Yes, it was.

Williangalvani, 2019-04-02:
I'll check what happens with the udev rules if a device matching the rules connects before the other one drops

jaxxzer, 2019-05-09:
This has come up as an issue with rebooting the autopilot generally. When the autopilot is rebooted, mavproxy _sometimes_ does not re-connect to the autopilot when it comes back up.

A curious thing is that refreshing the _system_ webpage will consistently re-establish the connection in mavproxy:

```
Device /dev/autopilot is dead
link 1 down
no link
no link
no link
no link
no link
no link
no link
no link
no link
no link
Device /dev/autopilot reopened OK
?CamPanTet]Flight battery warning
link 1 OK
heartbeat OK
```

I'm guessing this is related to udev and our /autopilot rules, but reloading rules in another terminal, or doing udevadm info (as the system page does) has no effect.

closed 2022-06-17 by ES-Alexander