bluerobotics/companion #251: Companion update button not appearing

jaxxzer, 2019-04-29:
When a user is on a version of companion prior to the latest stable release, an 'update avialable' button should appear on the _system_ page to update the companion software.

This button will only appear if the ROV has a connection to the internet, as indicated on the _network_ page.

If the button does not appear, try to refresh the page, then wait 10 seconds for the button to appear. If you still do not see the button, please, download a system log (using the button at the bottom of the _system_ page and share it here or in the discussion here: https://discuss.bluerobotics.com/t/no-companion-update-button-available/4061

This might be related to https://github.com/bluerobotics/companion/issues/247

jaxxzer, 2019-04-29:
If you continue to experience issues with this, and you [need](https://github.com/bluerobotics/companion/blob/master/release-notes.txt#L97) to update to 0.0.17, you will need to remove the sd card from the RASPBERRY PI inside of the ROV, and perform the [companion installation instructions](https://www.ardusub.com/getting-started/installation.html).

Williangalvani, 2020-05-05:
relevant log section from a user:

```
2020-04-29T18:54:41+0000 index.js:1336 (Socket.) get companion latest
2020-04-29T18:54:41+0000 index.js:1390 (Socket.) get ardusub version
2020-04-29T18:54:41+0000 index.js:1328 () null0.0.18

2020-04-29T18:54:42+0000 index.js:1338 () nullerror: tag ‘stable’ not found.
error: unable to resolve reference refs/tags/stable: Invalid argument
From https://github.com/bluerobotics/companion
! [new tag] stable -> stable (unable to update local ref)
warning: ignoring broken ref refs/tags/stable.
fatal: ambiguous argument ‘HEAD…refs/tags/stable’: unknown revision or path not in the working tree.
Use ‘–’ to separate paths from revisions, like this:
‘git […] – […]’
```

Williangalvani, 2020-05-05:
from https://discuss.bluerobotics.com/t/software-updates-april-2020/7234/21

jaxxzer, 2020-05-18:
@Williangalvani the user was on 0.0.18: https://github.com/bluerobotics/companion/blob/0.0.18/br-webui/index.js#L1337

there's some potential solutions proposed [here](https://stackoverflow.com/questions/10068640/git-error-on-git-pull-unable-to-update-local-ref), but the root of the issue remains unclear to me.

jaxxzer, 2020-05-18:
There is some better information on the problem [here](https://stackoverflow.com/questions/2998832/git-pull-fails-unable-to-resolve-reference-unable-to-update-local-ref), and I can think of two possibilities:

- There is an issue with the `stable` tag moving in the remote repository, and the local ref becomes desynchronized.
- The power was removed from the computer while it was writing to /refs/tags/stable, and the file is corrupt. I think this situation is more likely as `git tag -d stable` is failing with `error: tag ‘stable’ not found.`

PR #350 (2020-06-26): index: Remove lock files if they exist
The check for new versions fail since the lock file can exist

Fix #251
closed 2020-06-26 by patrickelectric