bluerobotics/companion #297: udp video broadcast causes issues

jaxxzer, 2020-01-14:
[This patch](https://github.com/bluerobotics/companion/commit/2ffa93b556b485326958c666436bf23413a094e5) works fine on my desktop with an ethernet cable to my router. It was presenting issues on ROVs with tether + fathom x in testing, though.

patrickelectric, 2020-01-14:
I did the setup here with the fathom-x and a 20MB/s Up-Down connection. the broadcast appears to be working without problems and it uses the same bandwidth as the normal fixed ip stream.
the max bandwidth from the video at 1920x1080@30fps is 400KB/s

jaxxzer, 2020-01-27:
We can revert this till we figure it out better. Then we can move forward with 0.0.19.

jaxxzer, 2020-01-28:
fixed: https://github.com/bluerobotics/companion/pull/302

closed 2020-01-28 by jaxxzer