bluerobotics/companion #322: v0.0.19 and 0.1.0: Does not connect to "Cube" type Autopilots

kklemens, 2020-03-23:
This isn't a huge issue for most of our customers, but while updating my personal BR2, I couldn't get a connection to either a Purple or Black Cube while I could with a 3DR Pixhawk1.

All autopilots were updated to 4.0.1.

Tested with 0.0.19 on a 3B and 0.1.0 on a 4B.

This may be related to the autoconnect issue with these types of boards (got fixed in QGC v4.0, but the drivers may not be in Companion): https://github.com/mavlink/qgroundcontrol/issues/7261

Williangalvani, 2020-05-05:
This is probably due to the way we now link the autopilots to /dev/autopilot.
as a workaround you can put the correct endpoint in http://192.168.2.2:2770/mavproxy

If it used to work, it is probably /dev/ttyACM0

petegreg, 2021-01-03:
Adding /dev/ttyACM0,115200 doesn't work for automatic connection to Cube Purple but the connection does work after mavproxy restarts as a result of attempting to update ardusub from the System page. The update doesn't work but at least you get the connection. Is the a shortcut to restart mavproxy from the command line? Restarting from the mavproxy page doesn't work.

proficnc, 2023-07-18:
Is this still a bug? Really? May want to mark it for a dev call.

patrickelectric, 2023-07-19:
Hi @proficnc, I hope that you are having a nice day. I believe that you are thinking that this repository is the same as [ArduPilot companion project](https://github.com/ArduPilot/companion). This is a Blue Robotics fork.

But, since you are here, I would highly recommend to take a look in our new companion software here: https://github.com/bluerobotics/BlueOS