bluerobotics/companion #325: wifi ap list is not populated correctly

jaxxzer, 2020-04-17:
It does not happen every time, it looks like maybe i'm getting some access point in range that causes the issue.

Looks like this commit (prior to 0.0.17 release) is to blame: https://github.com/bluerobotics/companion/commit/1cd1127043ec0628655971628a01450dd4e04fcf#diff-6d6e3e1d1bd652d84685e1d41dc7b7c4

[image]

jaxxzer, 2020-04-17:
does work
```
$ wpa_cli scan_results
Selected interface 'wlan0'
bssid / frequency / signal level / flags / ssid
48:5d:36:c9:44:b6       2462    -77     [WPA2-PSK-CCMP][WPS][ESS]       Fios-FWLN7
48:5d:36:c6:bd:d6       2412    -70     [WPA2-PSK-CCMP][WPS][ESS]       SuperSerbia
fa:54:b8:db:b8:a1       2412    -76     [WPA2-PSK-CCMP][WPS][ESS][P2P]  DIRECT-G1-FireTV_7a2e
84:17:ef:95:74:3e       2437    -65     [WPA2-PSK-CCMP][WPS][ESS]       MINCEY
20:c0:47:dc:89:a6       2412    -82     [WPA2-PSK-CCMP][WPS][ESS]       Fios-FWLN7
18:1b:eb:0f:e8:43       2437    -85     [WPA2-PSK-CCMP][WPS][ESS]       6MRH3
48:5d:36:30:16:de       2462    -86     [WPA2-PSK-CCMP][WPS][ESS]       FiOS-LU6PN
84:17:ef:95:74:41       2437    -68     [WPA2-PSK-CCMP][ESS]
84:17:ef:95:74:40       2437    -71     [ESS]   xfinitywifi
04:a2:22:d4:3d:69       2462    -84     [WPA2-PSK-CCMP][WPS][ESS]       Fios-tCg4C
00:7f:28:3a:94:11       2437    -92     [WPA2-PSK-CCMP][WPS][ESS]       ZSFG7
62:45:cb:1f:b1:80       2417    -87     [WPA2-PSK-CCMP][ESS]    RT-AC1200_80_2G_Guest
84:17:ef:95:74:43       2437    -67     [WPA-EAP-CCMP][WPA2-EAP-CCMP][ESS]
78:d2:94:19:34:ee       2462    -89     [WPA2-PSK-CCMP][WPS][ESS]       FiOS-9R390_2GEXT
20:c0:47:36:d4:e8       2462    -90     [WPA2-PSK-CCMP][WPS][ESS]       Fios-U4F59
72:f8:53:00:23:c3       2412    -87     [WPA2-PSK-CCMP][ESS]    Fios-nBh5W-Guest
b8:f8:53:00:23:c2       2412    -85     [WPA2-PSK-CCMP][WPS][ESS]       Fios-nBh5W
04:a2:22:ca:de:48       2437    -93     [WPA2-PSK-CCMP][WPS][ESS]       Fios-nBh5W
bc:9b:68:3a:d3:e1       2412    -86     [WPA2-PSK-CCMP][WPS][ESS]       Spider-LAN
62:a2:22:ca:de:49       2437    -91     [WPA2-PSK-CCMP][ESS]    Fios-nBh5W-Guest
62:a2:22:ca:c3:55       2462    -91     [WPA2-PSK-CCMP][ESS]    Fios-nBh5W-Guest
44:1c:12:93:61:25       2437    -92     [WPA-EAP-CCMP][WPA2-EAP-CCMP][ESS]
04:a2:22:ca:c3:54       2462    -86     [WPA2-PSK-CCMP][WPS][ESS]       Fios-nBh5W
60:45:cb:0f:b1:80       2417    -86     [WPA2-PSK-CCMP][WPS][ESS]       MACSey
fc:15:b4:42:0a:f2       2437    -94     [WPA2-PSK-CCMP][ESS]    HP-Print-F2-Deskjet 2540 series
94:6a:77:c2:4c:eb       2437    -89     [ESS]   xfinitywifi
00:19:9d:ff:c5:23       2437    -94     [ESS]   Bed room master
48:5d:36:b7:81:20       2462    -91     [WPA2-PSK-CCMP][WPS][ESS]       FiOS-9R390
bc:9b:68:3a:d3:e6       2412    -88     [WPA-EAP-CCMP][WPA2-EAP-CCMP][ESS]
c8:a7:0a:a2:c9:a4       2462    -92     [WPA2-PSK-CCMP][WPS][ESS]       FiOS-K42S7
c6:27:95:61:74:c0       2412    -92     [ESS]   xfinitywifi
```

does not work
```
$ wpa_cli scan_results
Selected interface 'wlan0'
bssid / frequency / signal level / flags / ssid
48:5d:36:c9:44:b6       2462    -75     [WPA2-PSK-CCMP][WPS][ESS]       Fios-FWLN7
48:5d:36:c6:bd:d6       2412    -67     [WPA2-PSK-CCMP][WPS][ESS]       SuperSerbia
fa:54:b8:db:b8:a1       2412    -76     [WPA2-PSK-CCMP][WPS][ESS][P2P]  DIRECT-G1-FireTV_7a2e
84:17:ef:95:74:3e       2437    -67     [WPA2-PSK-CCMP][WPS][ESS]       MINCEY
20:c0:47:dc:89:a6       2412    -81     [WPA2-PSK-CCMP][WPS][ESS]       Fios-FWLN7
18:1b:eb:0f:e8:43       2437    -89     [WPA2-PSK-CCMP][WPS][ESS]       6MRH3
48:5d:36:30:16:de       2462    -89     [WPA2-PSK-CCMP][WPS][ESS]       FiOS-LU6PN
84:17:ef:95:74:40       2437    -73     [ESS]   xfinitywifi
04:a2:22:d4:3d:69       2462    -88     [WPA2-PSK-CCMP][WPS][ESS]       Fios-tCg4C
00:7f:28:3a:94:11       2437    -92     [WPA2-PSK-CCMP][WPS][ESS]       ZSFG7
62:45:cb:1f:b1:80       2417    -84     [WPA2-PSK-CCMP][ESS]    RT-AC1200_80_2G_Guest
84:17:ef:95:74:43       2437    -68     [WPA-EAP-CCMP][WPA2-EAP-CCMP][ESS]
72:f8:53:00:23:c3       2412    -89     [WPA2-PSK-CCMP][ESS]    Fios-nBh5W-Guest
b8:f8:53:00:23:c2       2412    -88     [WPA2-PSK-CCMP][WPS][ESS]       Fios-nBh5W
04:a2:22:ca:de:48       2437    -94     [WPA2-PSK-CCMP][WPS][ESS]       Fios-nBh5W
62:a2:22:ca:c3:55       2462    -93     [WPA2-PSK-CCMP][ESS]    Fios-nBh5W-Guest
04:a2:22:ca:c3:54       2462    -88     [WPA2-PSK-CCMP][WPS][ESS]       Fios-nBh5W
60:45:cb:0f:b1:80       2417    -88     [WPA2-PSK-CCMP][WPS][ESS]       MACSey
48:5d:36:b7:81:20       2462    -91     [WPA2-PSK-CCMP][WPS][ESS]       FiOS-9R390
00:7f:28:6a:bb:18       2437    -96     [WPA2-PSK-CCMP][ESS]    W5TWC
fa:54:b8:db:b8:a1       2412    -77     [WPA2-PSK-CCMP][WPS][ESS][P2P]  \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
44:1c:12:93:61:23       2437    -90     [WPA2-PSK-CCMP][ESS]
20:c0:47:1e:85:a6       2412    -92     [WPA2-PSK-CCMP][WPS][ESS]       Fios-5GWR2
44:1c:12:93:61:25       2437    -91     [WPA-EAP-CCMP][WPA2-EAP-CCMP][ESS]
00:19:9d:ff:c5:23       2437    -99     [ESS]   Bed room master
84:17:ef:95:74:41       2437    -68     [WPA2-PSK-CCMP][ESS]
6c:b0:ce:bb:c5:b8       2427    -87     [WPA2-PSK-CCMP][WPS][ESS]       Wolf Creek Pass
78:d2:94:19:34:ee       2462    -90     [WPA2-PSK-CCMP][WPS][ESS]       FiOS-9R390_2GEXT
44:1c:12:93:61:20       2437    -91     [WPA2-PSK-CCMP][WPS][ESS]       Wifi2.45
```

jaxxzer, 2020-04-17:
after playing, it looks like this one is to blame
`fa:54:b8:db:b8:a1       2412    -77     [WPA2-PSK-CCMP][WPS][ESS][P2P]  \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00`

jaxxzer, 2020-07-30:
@dgudiel reports this behavior occurs on 0.0.21

patrickelectric, 2020-07-30:
This was fixed in master, why the issues is open ? The new wifi manager was not release in 0.21

patrickelectric, 2020-08-04:
Can this be closed ?

PR #333 (2020-06-26): Add wifi service
This PR also allows utf-8 chars in ssid

[image]

Fix #325
closed 2020-08-04 by jaxxzer