bluerobotics/companion #357: ping360_bridge_manager

1chick, 2020-07-02:
> def device_has_screen(device):

>         output = subprocess.check_output(["sudo", "-Au", "pi", "screen", "-ls"], universal_newlines=True)
>         return output.decode()

can i comment  _return output.decode()_ ?

Williangalvani, 2020-07-10:
Yes, that sound like a leftover of debugging. Are you using a screen version newer than 4.2?

1chick, 2020-07-14:
> Yes, that sound like a leftover of debugging. Are you using a screen version newer than 4.2?

ok，Thanks.  No its 4.06, i can't get normal output if the code is not commented. Would you test it as you have free time?

Williangalvani, 2020-07-14:
> ok，Thanks. No its 4.06, i can't get normal output if the code is not commented. Would you test it as you have free time?

Yes, I  reopened the issue to take a better look at it later.

closed 2022-06-17 by ES-Alexander