bluerobotics/companion #391: Water Linked Driver Connection

brianhBR, 2021-01-21:
While testing the new Water Linked UGPS G2 system it was noticed that the companion water linked service wouldn't connect automatically. 

The fix was to log in to the water linked page in companion and click restart. This was necessary each time the water linked system was restarted. 

I'm planning to collect more information to help identify the issue next time we are out on a dive.

BoKoppel, 2021-03-24:
Hi!
I have a system with BR and Waterlinked UGPS G2 that could be used for testing this issue.
Testing by me or by BR via internet.
/ Bo

Williangalvani, 2021-05-11:
Hi. It seems that the UGPS G2 is not being advertised at the usual "waterlinked.local" name. When the service is restarted. it is restarted with "192.168.2.94" and that is why it works.

PR #400 (2021-08-05): Underwater GPS tweaks
Note that changing the starting ip to 192.168.2.94 __could__ break some unorthodox setups, but it will fix the main issue with Underwater GPS G2.

As I don't have the testing setup ready, I don't feel comfortable doing larger changes in the codebase to handle both mDNS addresses.

fix #391
closed 2021-08-05 by jaxxzer