bluerobotics/companion #420: ping360 ethernet support

jaxxzer, 2021-09-02:
Reports that ping360 is not automatically connecting with companion when connected by ethernet.
investigate.

patrickelectric, 2021-09-03:
If Ping360 is configured with ethernet connection internally, it should work with a normal router or a switch.
What can happen is for Ping-Viewer to detect Ping360 through a switch but what you may see a “yellow circle” in Ping-Viewer saying that something may be wrong, and if you hover the mouse over it a message you inform that the network subnet is not compatible. 
It should be fixed if the user enable the DHCP server on companion or configure Ping360 static IP address before connecting.
What we could do is to broadcast a `setStaticIPAddress` message but this may brake any case where multiple sonars/compatible sensors are in the network.

patrickelectric, 2021-09-03:
But you'll see Ping-Viewer detecting the sonar, no matter what. If Ping-Viewer does not detect it, it's probably a hardware issue.

dgudiel, 2021-09-04:
I tested the Ping360 connection with companion in DHCP Server mode and I got the following result:
[image]

 So even with companion in DHCP server mode the device wasn't configured with a correct IP address on a compatible subnet. I attempted to set a static address by clicking the  gear icon but the settings would not save after multiple attempts.