bluerobotics/companion #426: Install the firmware on raspberry pi 4 problem

fjosen, 2021-11-11:
hello.
Can some one tell me how i can install the firmware on the raspberry pi 4.  Do i need to install rapsberry pi (full) or flash the card whit the firmware. from this site: https://www.ardusub.com/quick-start/installing-companion.html

rafaellehmkuhl, 2021-11-12:
Raspberry Pi 4 boards are not supported on currently version of Companion, but we are developing a brand-new version from scratch, which supports Pi4 boards and aims to be much more user-friendly. If you want to try it, you can read more about [here](https://docs.bluerobotics.com/ardusub-zola/software/companion/1.0/installation/).

fjosen, 2021-11-16:
Tanks. I have tried this version, i open it on chrome and it works. but i cudnt connect to the groundcontrol (ardupilot). i havent received the pixhawk yet, so meybe i need this before it Will connect.

I have one raspberry 3b+. i have tried this https://www.ardusub.com/quick-start/installing-companion.html  but the powerlight on the raspberry pi 3B+ blinking (red).  what is wrong?

rafaellehmkuhl, 2021-11-16:
> Tanks. I have tried this version, i open it on chrome and it works. but i cudnt connect to the groundcontrol (ardupilot). i havent received the pixhawk yet, so meybe i need this before it Will connect.

Regarding this, did you tried the QGC as a client (using `companion.local` on the IP)? We changed the behaviour from old companion to the new one.

> I have one raspberry 3b+. i have tried this https://www.ardusub.com/quick-start/installing-companion.html but the powerlight on the raspberry pi 3B+ blinking (red). what is wrong?

@ES-Alexander do you knoe something about that? If i remember correctly this is usually lack of wifi configuration.

patrickelectric, 2021-11-16:
> Tanks. I have tried this version, i open it on chrome and it works. but i cudnt connect to the groundcontrol (ardupilot). i havent received the pixhawk yet, so meybe i need this before it Will [...]

I'll now have a QGC connection if you don't have a flight controller, like a pixhawk

closed 2022-06-17 by ES-Alexander